Inxton/TcOpen #26: [NEW-FEATURE] TcoDialogs localisation support and permissions support

KrisztianOlah, 2025-06-10:
**Is your feature request related to a problem? Please describe.**
TcoCustomDialog does not currently support localising the push buttons, and they cannot be hid or disabled when the user lacks permission to excecute a certain action.

**Describe the solution you'd like**
We would need to implement both to aid speedy development, because currently it is easier to write custom solutions particularly when localisation is required.

**Describe alternatives you've considered**
Multiple dialogs based on user and language, but too many options.

peterbarancek, 2025-09-01:
/cib

peterbarancek, 2025-09-05:
Hi,  
the language mutations has been completed.  
If you would like to use permissions, here is an example of how to handle it directly from the PLC:

Note: The first part (in comments) is required if a dialog has already appeared and the user level changes at that moment.

 ```

///////////////////////////////////////////////////////////////////
        // Trigger on user level change
        rTrig(CLK := _userLevel <> _lastLevel);

        IF rTrig.Q THEN
            _dialogCustomized.Restore(); // Close/hide dialog on change
            tHide(IN := FALSE); // Reset timer
        END_IF

        // Start/extend timer when a change is detected
        tHide(IN := rTrig.Q OR tHide.IN, PT := T#100MS); // Delay before showing dialog again

        // While timer is active (counting but not yet elapsed), block dialog
        IF NOT tHide.Q AND tHide.IN THEN
            RETURN;
        END_IF

        ////////////////////////////////////////////////////////////////
        IF _userLevel = 'OperatorGroup' THEN
            _lastLevel := _userLevel;
            _dialogCustomized.Show()
                .WithType(eDialogType.Info)
                .WithOption1('<#This is Option1=>retry#>')
                .WithOption2('<#This is Option2=>first step#>')
                .WithOption3('<#This is Option3 =>continue#>')
                .WithCaption('<#Hey#>')
                .WithText('<#What we are going to do.#>');
        ELSE
            _lastLevel := _userLevel;
            _dialogCustomized.Show()
                .WithType(eDialogType.Info)
                .WithOption1('<#This is Option1=>retry#>')
                .WithOption2('<#This is Option2=>first step#>')
                .WithOption3('<#This is Option3 =>continue#>')
                .WithOption4('<#This is Option4=>stop sequence#>')
                .WithCaption('<#Hey#>')
                .WithText('<#What we are going to do.#>');
        END_IF;

        IF (_dialogCustomized.Answer = TcoCore.eCustomizedDialogAnswer.Option1) THEN
            _state := 50;
        END_IF;

        IF (_dialogCustomized.Answer = TcoCore.eCustomizedDialogAnswer.Option2) THEN
            _state := 0;
        END_IF;

        IF (_dialogCustomized.Answer = TcoCore.eCustomizedDialogAnswer.Option3) THEN
            _state := 51;
        END_IF;

        IF (_dialogCustomized.Answer = TcoCore.eCustomizedDialogAnswer.Option4) THEN
            _state := 0;
            _invokeCustomizedDiaglog := FALSE;
            _invokeDiaglog1 := FALSE;
        END_IF;
```

PR #36 (2025-09-08): [NEW-FEATURE] TcoDialogs localisation support and permissions support
closes #26
closed 2025-09-08 by peterbarancek